feelfreelinux/octo4a #528: Install fails on Sony Z1c LineageOS: proot signal 11

prullmann, 2025-05-07:
I am trying to get octo4a running on my old Sony Xperia Z1 compact (broken touch).
Octo4a failed to install on stock Android 4.4.4 and since Lineage OS existed, I didn't try stock 5.1.1.
But also on LineageOS 18, 17 and 15 I had no luck.
I tried octo4a 2.1.0, but also 2.0.0 and 1.2.6 just in case.

On LineageOS 15.1 (Android 8.1) using octo4a 2.1.0 I dug a bit deeper:
Installation fails right after downloading bootstrap with the error: `proot info: vpid 1: terminated with signal 11`
See also similar error in #255 

I can confirm the error using adb shell:
```
% adb root
restarting adbd as root
% adb shell                         
amami:/ # cd /data/data/com.octo4a/files/bootstrap
amami:/data/data/com.octo4a/files/bootstrap # ./entrypoint.sh octoprint cat /etc/motd                                                                                                
proot warning: can't sanitize binding "/apex": No such file or directory
proot info: vpid 1: terminated with signal 11
```

It sounds very much like this issue in proot: https://github.com/termux/proot/issues/107
There it was fixed with a single-line commit: https://github.com/termux/proot/commit/6356be2321a6caf2916cecdbae80dde84addecda

element73, 2025-06-08:
> It seems this upstream fix was already included in the latest [octo4a-bootstrap-builder v073b9017-1.10.3](https://github.com/feelfreelinux/octo4a-bootstrap-builder/releases/tag/v073b9017-1.10.3).
> 
> So back to square one...

Were you able to fix the issue? I'm getting this exact problem on an old android tablet.

spacekomet, 2025-06-14:
Solution for error "proot info: vpid 1: terminated with signal 11" for me was this https://github.com/termux/proot/issues/119#issuecomment-687705444.

After failed octoprint installation, I used adb as a root and set the environment variable PROOT_NO_SECCOMP=1 in  /data/data/com.octo4a/files/bootstrap/entrypoint.sh. I also added the environment: export EXTRA_BIND="-b /data/data/com.octo4a/files:/mnt/external -b /data/data/com.octo4a/files/serialpipe:/dev/ttyOcto4a", because in my case apk app  cannot access /storage/emulated/0/Android/data/com.octo4a/Files where it normally saves the config.yaml, although I granted permission for it. Variable EXTRA_BIND is otherwise hardcoded in the apk. 

However this only solved this one error. 
Then i got this https://github.com/feelfreelinux/octo4a/issues/515. Unfortunately I don't really know why this error occurs when running the apk app, because I can manually run the octroprint (as a root) from adb with entrypoint.sh or from inside proot without issues.  (I  assume it has something to do with the permissions though).

I'm on lineageOS 19 (android 12), Samsung S2 I9100. 

I've tried it on 2 other "brand new" :) phones, where installation worked and octoprint runs, but they both don't support USB OTG. 
- lenovo A600 (lineageOS 22.2, Android 15) no hardware support for OTG
- Lumia 950 (lineageOS 18.1, Android 11), no linux drivers :).

prullmann, 2025-06-25:
> Were you able to fix the issue? I'm getting this exact problem on an old android tablet.

No, I now found a different phone and that worked immediately.
I should have given up a lot earlier I guess...